up-division/pinctrl-upboard #3: PINCTRL ASL for UP2

lzala, 2023-04-07:
I see that  [ acpi/gpio/pinctrl-up-ehl.asl ](https://github.com/up-division/pinctrl-upboard/blob/master/acpi/gpio/pinctrl-up-ehl.asl) is available for UPN-EHL01/UPS-EHL01. The device I'm using is UP-APL01 and SSDT is already available. I was able to dump and decompile it to:
```
DefinitionBlock ("", "SSDT", 1, "AAEON", "PINCTRL", 0x00000001)
{
    External (_SB_.GPO0, DeviceObj)
    External (_SB_.GPO1, DeviceObj)
    External (_SB_.GPO2, DeviceObj)
    External (_SB_.GPO3, DeviceObj)
    External (ABID, IntObj)
    External (OSSL, IntObj)

    Scope (\_SB)
    {
        Device (PCTL)
        {
            Name (_HID, "AANT0F01")  // _HID: Hardware ID
            Name (_DDN, "UP Squared FPGA-based Pin Controller")  // _DDN: DOS Device Name
....
```

But I'm interested it the .asl is available anywhere online.
The only mention I could find is from a kernel commit:
https://patchwork.ozlabs.org/project/linux-gpio/patch/20180421083008.27411-4-javier@emutex.com/

Is it PINCTRL SSDT part of the BIOS FW? I did not figure it out how the ACPI table/overlays are shared between fw/bootloader/kernel. 

BTW any plans to officialize pinctrl-upboard as a .deb package until it will be probably integrated in the kernel?

Thanks

gary0124, 2023-04-27:
EHL is not included PINCTRL ACPI declare for pinctrl-upboard driver, so we injection when driver installation, 
you can dump PINCTRL ACPI in other boards. 
upstream is on going now.

lzala, 2023-04-27:
thanks for he info, I think we can close the ticket

gary0124, 2023-04-28:
upstream is on going

closed 2023-04-28 by gary0124