up-division/pinctrl-upboard #6: Unable to use version 1.0.8 on Ubuntu 22

e10eqs, 2023-06-15:
Hello, I'm trying to get gpio working on the UP Squared board with linux 22.04 on kernel version 5.15.72 using the pinctrl package (version 1.0.8). After installing the package according to the directions [here](https://github.com/up-division/pinctrl-upboard/tree/v1.0.8) I still cannot communicate with the gpio pins using gpioset/gpioget in a bash terminal or using the python lgpio library. I've also had a look at the gpio support [wiki](https://github.com/up-board/up-community/wiki/40Pin-Header) which states that this functionality is supported with my linux/kernel versions.  When I run dkms status I get an error message that says "pinctrl-upboard/1.0.8, 5.15.0-73-generic, x86_64: installed (WARNING! Diff between build and installed module!)". I'm hoping the developers can publish more docs or address underlying issues. Thanks!

gary0124, 2023-06-30:
22.04 with kernel 5.15.72 is no problem, you can post in UP Community to get support 
https://forum.up-community.org/?utm_source=up-board_homepage&utm_medium=web&utm_campaign=community_upboard_nav

e10eqs, 2023-06-30:
Hello,

I posted <https://forum.up-community.org/discussion/5187/gpio-on-up-squared#latest> in the forum a while ago and am still waiting on a response. If there are any more resources I’m not seeing or documentation you could point me to that would be greatly appreciated.

-Ethan

gary0124, 2023-07-04:
using sysfs and uAPIV2 to access GPIO is no problem, it's gpiod prlboem.

closed 2023-07-04 by gary0124

e10eqs, 2023-07-04:
Hi Gary, I haven’t been using uAPIV2, what is that? Do you have a link to some documentation that I could look at?Thanks!-Ethan

gary0124, 2023-07-04:
it's linux kernel provided user API V2, you can check & build kerenl tools to testing, https://github.com/torvalds/linux/tree/master/tools/gpio