up-division/pinctrl-upboard #8: .deb package installatin fails on Debian Bookworm Up Squared 7000 Pro board

Plopli, 2023-09-08:
pinctrl-upboard_1.1.1_all.deb package

The installation enter a infinite loop with messages:
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/pinctrl-upboard/1.1.1/source/dkms.conf)

The installer seems to span process indefinitely. I did not manage to capture what exactly the process is.

Plopli, 2023-09-10:
REMAKE_INITRD is not the problems.
The build trigger an infinite spawn of processes that looks like these:
root        1474     767  0 09:51 pts/0    00:00:00           sudo dpkg -i pinctrl-upboard_1.1.1_all.deb
root        1475    1474  0 09:51 pts/1    00:00:00             sudo dpkg -i pinctrl-upboard_1.1.1_all.deb
root        1476    1475  0 09:51 pts/1    00:00:00               dpkg -i pinctrl-upboard_1.1.1_all.deb
root        1488    1476  0 09:51 pts/1    00:00:00                 /bin/sh /var/lib/dpkg/info/pinctrl-upboard.postinst configure
root        1489    1488  0 09:51 pts/1    00:00:00                   /usr/bin/perl -w /usr/share/debconf/frontend /usr/lib/dkms/common.postinst pinctrl-upboard 1.1.1 /usr/share/pinctrl-upboard-dkms
root        1497    1489  0 09:51 pts/1    00:00:00                     /bin/sh /usr/lib/dkms/common.postinst pinctrl-upboard 1.1.1 /usr/share/pinctrl-upboard-dkms
root        2788    1497  0 09:51 pts/1    00:00:00                       /bin/bash /usr/sbin/dkms install -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        2823    2788  0 09:51 pts/1    00:00:00                         /bin/bash /usr/sbin/dkms install -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        3041    2823  0 09:51 pts/1    00:00:00                           /bin/sh /var/lib/dkms/pinctrl-upboard/1.1.1/source/setup.sh pinctrl-upboard 1.1.1
root        3101    3041  0 09:51 pts/1    00:00:00                             /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        3136    3101  0 09:51 pts/1    00:00:00                               /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        3468    3136  0 09:51 pts/1    00:00:00                                 /bin/sh /var/lib/dkms/pinctrl-upboard/1.1.1/source/setup.sh pinctrl-upboard 1.1.1
root        3528    3468  0 09:51 pts/1    00:00:00                                   /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        3563    3528  0 09:51 pts/1    00:00:00                                     /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        3903    3563  0 09:51 pts/1    00:00:00                                       /bin/sh /var/lib/dkms/pinctrl-upboard/1.1.1/source/setup.sh pinctrl-upboard 1.1.1
root        3963    3903  0 09:51 pts/1    00:00:00                                         /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        3998    3963  0 09:51 pts/1    00:00:00                                           /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        4338    3998  0 09:51 pts/1    00:00:00                                             /bin/sh /var/lib/dkms/pinctrl-upboard/1.1.1/source/setup.sh pinctrl-upboard 1.1.1
root        4398    4338  0 09:51 pts/1    00:00:00                                               /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        4433    4398  0 09:51 pts/1    00:00:00                                                 /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        4773    4433  0 09:51 pts/1    00:00:00                                                   /bin/sh /var/lib/dkms/pinctrl-upboard/1.1.1/source/setup.sh pinctrl-upboard 1.1.1
root        4833    4773  0 09:51 pts/1    00:00:00                                                     /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        4868    4833  0 09:51 pts/1    00:00:00                                                       /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        5208    4868  0 09:51 pts/1    00:00:00                                                         /bin/sh /var/lib/dkms/pinctrl-upboard/1.1.1/source/setup.sh pinctrl-upboard 1.1.1
root        5268    5208  0 09:51 pts/1    00:00:00                                                           /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        5303    5268  0 09:51 pts/1    00:00:00                                                             /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        5643    5303  0 09:51 pts/1    00:00:00                                                               /bin/sh /var/lib/dkms/pinctrl-upboard/1.1.1/source/setup.sh pinctrl-upboard 1.1.1
root        5703    5643  0 09:51 pts/1    00:00:00                                                                 /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        5738    5703  0 09:51 pts/1    00:00:00                                                                   /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
root        6078    5738  0 09:51 pts/1    00:00:00                                                                     /bin/sh /var/lib/dkms/pinctrl-upboard/1.1.1/source/setup.sh pinctrl-upboard 1.1.1
root        6138    6078  0 09:51 pts/1    00:00:00                                                                       /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.1.0-11-amd64
[... cut ...]
root       11393   11358  0 09:51 pts/1    00:00:00                                                                                                                                                 /bin/bash /usr/sbin/dkms install --force -m pinctrl-upboard -v 1.1.1 -k 6.
root       11733   11393  0 09:51 pts/1    00:00:00                                                                                                                                                   /bin/sh /var/lib/dkms/pinctrl-upb

gary0124, 2023-09-14:
fix in v1.1.2

closed 2023-09-14 by gary0124

Plopli, 2023-09-14:
I confirm build is working now.