up-division/pinctrl-upboard #10: GPIO not working on 1.1.3 UP 7000 (N100)

Climax777, 2024-01-12:
uname -a:
```
Linux jericho-up 5.15.0-91-generic #101~20.04.1-Ubuntu SMP Thu Nov 16 14:22:28 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

I see this in my dmesg.
```
[    5.104753] upboard_cpld: loading out-of-tree module taints kernel.
[    5.104786] upboard_cpld: module verification failed: signature and/or required key missing - tainting kernel
[    5.105115] upboard-cpld AANT0F04:00: Failed to initialize FPGA common GPIOs: -517
```

I have install pinctrl-upboard 1.1.3. I don't see any active modules, but the LEDs are exported in /sys/class/leds
However, writing to the brightness has no effect. I'm going to test this again with a fresh installed ubuntu 20 with kernel 5.15 and the pinctrl driver. Exporting gpios (27 for example) fails with:

```
[  435.859956] export_store: invalid GPIO 27
```

Please let me know if you need additional information.

Climax777, 2024-01-12:
I can confirm a fresh and clean ubuntu 20.04 install does exactly the same...

gary0124, 2024-03-22:
kernel version is too old for ADLN.
please use kernel version 5.19+ that included all built-in driver for ADLN.

closed 2024-03-22 by gary0124