up-division/pinctrl-upboard #12: Openwrt support

0x0B0x11, 2024-07-18:
Installing modules on openwrt ends with an error

0x0B0x11, 2024-07-18:
Openwrt x86_64

gary0124, 2024-07-22:
could you provide error log or screenshot when build error on OpenWRT x86_64?

0x0B0x11, 2024-07-30:
`root@78a5:~/pinctrl-upboard/files# make all
checking pwm prototypes...
rm -f ./protos.h
rm -f ./protos.o
sed -i 's/#define CHECK_SPI_SSP/#define CHECK_PWM/g' protos.c
make -i -C /lib/modules/6.1.89/build M=/root/pinctrl-upboard/files protos.o; ([ -f protos.o ] && echo "#define TYPES_NO_ERROR_CODE 1" > protos.h || echo "#define TYPES_NO_ERROR_CODE 0" > protos.h )
make[1]: Entering directory '/lib/modules/6.1.89/build'
make[1]: *** No rule to make target 'protos.o'.  Stop.
make[1]: Leaving directory '/lib/modules/6.1.89/build'
checking gpio prototypes...
rm -f ./protos.o
sed -i 's/#define CHECK_PWM/#define CHECK_GPIO/g' protos.c
make -i -C /lib/modules/6.1.89/build M=/root/pinctrl-upboard/files protos.o; ([ -f protos.o ] && echo "#define TYPES_NO_OFFSET 1" >> protos.h || echo "#define TYPES_NO_OFFSET 0" >> protos.h )
make[1]: Entering directory '/lib/modules/6.1.89/build'
make[1]: *** No rule to make target 'protos.o'.  Stop.
make[1]: Leaving directory '/lib/modules/6.1.89/build'
checking spi naming...
rm -f ./protos.o
sed -i 's/#define CHECK_GPIO/#define CHECK_SPI_NAMING/g' protos.c
make -i -C /lib/modules/6.1.89/build M=/root/pinctrl-upboard/files protos.o; ([ -f protos.o ] && echo "#define TYPES_IS_SLAVE 1" >> protos.h || echo "#define TYPES_IS_SLAVE 0" >> protos.h )
make[1]: Entering directory '/lib/modules/6.1.89/build'
make[1]: *** No rule to make target 'protos.o'.  Stop.
make[1]: Leaving directory '/lib/modules/6.1.89/build'
checking spi ssp...
rm -f ./protos.o
sed -i 's/#define CHECK_SPI_NAMING/#define CHECK_SPI_SSP/g' protos.c
make -i -C /lib/modules/6.1.89/build M=/root/pinctrl-upboard/files protos.o; ([ -f protos.o ] && echo "#define TYPES_SSP_PDEV 1" >> protos.h || echo "#define TYPES_SSP_PDEV 0" >> protos.h )
make[1]: Entering directory '/lib/modules/6.1.89/build'
make[1]: *** No rule to make target 'protos.o'.  Stop.
make[1]: Leaving directory '/lib/modules/6.1.89/build'
make -C /lib/modules/6.1.89/build M=/root/pinctrl-upboard/files modules
make[1]: Entering directory '/lib/modules/6.1.89/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/6.1.89/build'
make: *** [Makefile:30: all] Error 2`

gary0124, 2024-07-31:
looks it's not included packages for kernel moudle build in your enviroment, did you build it in openWRT?  or build it in your other build machine?

0x0B0x11, 2024-08-10:
I realized my mistake, I need to add a module when building openwrt, only it remains to write the correct MakeFile

0x0B0x11, 2024-08-20:
Could you help me write a makefile? I don't understand how to transfer everything you have in makefile and setup.sh to openwrt format

gary0124, 2024-10-07:
sorry, openwrt is not official supported OS.

closed 2024-10-07 by gary0124

ByteSystemOne, 2024-12-05:
> Или скажите, какие пакеты утомительно собирать напрямую в openwrt

hey, did you solve it?

ByteSystemOne, 2024-12-05:
> sorry, openwrt is not official supported OS.

But maybe you can help to add it? Not official but its will be good